YetiTool/easycut-smartbench #54: Raise the Spindle

LettyYeti, 2019-03-14:
If spindle is lowered and you want to move to the job's location, SB will not raise the spindle and in some cases it can cause cutter to catch on material.

[image]

[image]

LettyYeti, 2019-03-28:
@Neex101 

For solution to this: is it ok to just add m.zUP as the first command for these buttons?

Neex101, 2019-03-28:
Yep.
I worry that this safety net will annoy pro users. But we'll cross that bridge when we come to it. I'm thinking we'd demand a second button to be pushed simulataneously (like using the shift key) to allow an XY move without a zUp.

closed 2019-04-06 by LettyYeti