YetiTool/easycut-smartbench #60: Disable double tap for move button commands

Neex101, 2019-03-21:


LettyYeti, 2019-03-28:
@Neex101 more info needed on this one please

Neex101, 2019-03-28:
On manual move screen, up/down/left/right buttons... double tapping the button locks the button down, yielding an unexpectedly long move. Danger danger.

LettyYeti, 2019-04-16:
Kivy config file settings

LettyYeti, 2019-05-06:
Needs to be fixed within GRBL settings:

" a GRBL cancel Jog command (x85) will only work if it's not already cancelling a jog. So if you send the following in quick succession "Jog, Cancel, Jog, Cancel", the second cancel won't register whilst the first jog is in progress. It seems, however, the the second Jog is maybe added after the first cancel has flushed the buffer, so then the second Jog still happens..."

LettyYeti, 2019-10-01:
@Neex101 did the FW update help with this at all? Can't remember

Neex101, 2019-10-01:
Nope, double tap is alive and well, even on latest f/w :-(

closed 2022-02-07 by LettyYeti