YetiTool/easycut-smartbench #89: Deactivate/clear redundant buttons

Neex101, 2019-03-30:
E.g. need a coming soon for the two other apps seen in the lobby & deactivate scond one (it's live to a shape generating screen which skippy didn't finish)
Dev area needs sectioning off
There's a "template" button on the job screen (page symbol with a star on it) which leads to some unfinished tempalte dialogue
... and probz more items

LettyYeti, 2019-03-30:
Have sorted the template and lobby buttons. 

Dev area has a separate Git issue so will section that off. 

Have added question label bc need to know whether there is other stuff that needs addressing.

LettyYeti, 2019-04-07:
VIEW BUTTONS ON VIRTUAL BED

LettyYeti, 2019-10-01:
@Neex101 what are the view buttons meant to do - should I add a new ticket to add in functionality for them or just ditch them completely for now?

Neex101, 2019-10-01:
Ditch

closed 2019-10-01 by Neex101