YetiTool/easycut-smartbench #103: Unable to parse "Door" status

Neex101, 2019-04-05:
`ERROR status parse: Status invalid: <Door:0|MPos:0.000,0.000,-15.000|Bf:35,255|FS:0,0|Pn:XYZ>`
(Tested on Eclipse/Arduino mega)

Neex101, 2019-04-05:
Critical because of unknown side effects during loss of status

LettyYeti, 2019-04-06:
I don't think this is a problem on Benchy or on Riggy Pop, just Ardy.

LettyYeti, 2019-04-07:
update: is also a problem on Riggy Pop.

Neex101, 2019-04-07:
Definitely a problem on benchy

-- 

S i S
*Ed Sells* • *Director*
T • +44 1275 390721  M • +44 7833 478878
E • ed.sells@yetitool.com  W • www.yetitool.com
A • Unit 1 Westfield Court | Barns Ground | Clevedon | Bristol | BS21 6FQ |
UK
[image: Follow us on Facebook.] <https://www.facebook.com/yetitool/> [image:
Follow me us Twitter!] <https://twitter.com/yetitool> [image: Follow us on
LinkedIn!]  <https://www.linkedin.com/company/yeti-tool-ltd/>[image: See
our Yeti TV channel on YouTube.]
<https://www.youtube.com/channel/UC9dWz-Uk1K3ewhxLDG7yjtQ>
The e-mail and any files transmitted with it are confidential and may be
privileged, and are intended solely for the use of the individual or entity
to whom they are addressed. Any unauthorised direct or indirect
dissemination, distribution or copying of this message and any attachments
is strictly prohibited.

Neex101, 2019-04-07:
Haven't looked at all, but probz becuase can be either `Door:1` or `Door:0`

-- 

S i S
*Ed Sells* • *Director*
T • +44 1275 390721  M • +44 7833 478878
E • ed.sells@yetitool.com  W • www.yetitool.com
A • Unit 1 Westfield Court | Barns Ground | Clevedon | Bristol | BS21 6FQ |
UK
[image: Follow us on Facebook.] <https://www.facebook.com/yetitool/> [image:
Follow me us Twitter!] <https://twitter.com/yetitool> [image: Follow us on
LinkedIn!]  <https://www.linkedin.com/company/yeti-tool-ltd/>[image: See
our Yeti TV channel on YouTube.]
<https://www.youtube.com/channel/UC9dWz-Uk1K3ewhxLDG7yjtQ>
The e-mail and any files transmitted with it are confidential and may be
privileged, and are intended solely for the use of the individual or entity
to whom they are addressed. Any unauthorised direct or indirect
dissemination, distribution or copying of this message and any attachments
is strictly prohibited.

Neex101, 2019-04-12:
OK, so this is a hangover from Skippy code, and can simply be resolved (as per `door_stop_fix` branch).
No side-effects, so non critical.
Renaming issue to reflect that we need a screen for when the 'Door' status is activated, (so user doesn't have to manually reset and unlock each time) which will be a product of issue #123

closed 2020-03-21 by LettyYeti