YetiTool/easycut-smartbench #112: On checking file, after alarm, then exit file continues to stream

Neex101, 2019-04-05:
CRITICAL
If machine then gets reset, check mode gets reset and the machine can start jobbing

LettyYeti, 2019-04-05:
Do not understand??

Neex101, 2019-04-05:
Description updated

LettyYeti, 2019-04-07:
Ok so tried an if job in bounds when opening file checking. Does not work. In eclipse+arduino system I always get an error to say job is outside bounds on the file check screen now (inc. for test brick). On test rig, it doesn't think it's outside bounds and xyz_endurance still ends up with alarm raising straight away

Ways around this:

1. Accept alarm state and now that seq. streaming is default cancelled on soft_reset, no more ghost machine anyway. Can also add a safety "cancel seq streaming" or seq_streaming = False on leaving the screen if needsbe,

2. Go back to trying to suppress alarm screens, handle the alarm state automatically, and feedbacking to the user alongside error messages (inc. possible residual error 9s, unless they are explicitly excluded)

@Neex101 what works for you?

closed 2019-04-16 by LettyYeti