YetiTool/easycut-smartbench #123: Stop/pause status action review

Neex101, 2019-04-12:
We have various stop and pause contexts
What do we want to do for each? How should the machine react? How should the user resolve? What processes should we ensure are terminated?
Let's map out and understand thoroughly before we implement.

LettyYeti, 2019-04-16:
Handle door function from GRBL, without fucking up play pause etc. normal door state

LettyYeti, 2019-12-16:
If alarm state cancel/pauses job what happens: 

"Towards the end of a routing program the SmartBench stopped with an alarm.  Unable to go to the "Move Screen" in the alarm state, the only possible action was to cancel the job.  After canceling the job and navigating to the Move Screen, all of the move functions were non-functional.  The router bit was still engage in the work piece. The display didn't show an alarm; however, the red LEDs were flashing.  Attempting to clear the alarm didn't work.  Looking at the GCODE, there was an error message to "Check the Door",  It also seem that the move requests were shown in the GCODE with out any movement occurring.  Powering down and restarting the unit didn't help.  Where is the "door" that is causing this problem?" 

- try to replicate this, see what's happening

closed 2020-07-09 by LettyYeti