YetiTool/easycut-smartbench #125: Add "on limit switch" detection to enable homing function

Neex101, 2019-04-12:
Sometimes an axis can already be on a limit switch when the machine is asked to home.
This causes an alarm.
So the machine cannot home. So the alarm persists. So the machine cannot home etc etc etc.
Proposed solutions:
On entering homing function:
Pre-check grbl status to examine if any limits are arleady  triggered (shown in the `Pn:` section of the status). If found, move 20 mm in the direction that would untrigger the switch. THEN continue with homing function.
Pn: x = min x limit switch triggered
Pn: X = max x limit switch triggered
Pn: y = min y limit switch triggered
Pn: Y = max Y limit switch triggered
Pn: z = min z limit switch triggered

LettyYeti, 2019-05-04:
I'm not seeing this error on the test rig either? I can have the axes right up on the limit switches and homing still runs fine, no alarms seen?

Neex101, 2019-05-05:
We need to sync your version of grbl

LettyYeti, 2019-10-01:
@Neex101 can we close this?

Neex101, 2019-10-01:
Brings back some horrible memories :-) Ditch!

Neex101, 2019-10-01:
(no, done!)

closed 2019-10-01 by Neex101