YetiTool/easycut-smartbench #126: Optimise logging

Neex101, 2019-04-12:
From @Klazomenai :
We should have two modes of logging. Error/warning only for the wild and debug for the dev benches. Debugging mode should also be able to be turned on or off from the console. That will behave nicely, the only problem being that you won't have debug messages for the wild for if a specific cut will break. Debug mode will need to be turned on an another equal cut run to get the debug messages. But generally, debug logging is avoided in production.

Neex101, 2019-04-12:
Specific reference to the grbl status logs which are currently @ 10Hz

Neex101, 2019-04-16:
Possible cause of slowdown

LettyYeti, 2020-04-01:
send Ed sample logs

Neex101, 2020-04-12:
Cleaned up

Neex101, 2020-04-12:
Ah, wrong thread

LettyYeti, 2020-07-31:
@Neex101 can I assume this is covered by Job Logging in trello?

Neex101, 2020-08-01:
Yep

closed 2020-08-07 by LettyYeti