YetiTool/easycut-smartbench #211: USB "repair drive" bug

LettyYeti, 2019-11-05:
From customer - Whenever I reattached my USB drive to my computer AFTER having it attached to the SmartBench, I encounter a "repair drive" prompt from my computer. I would run the repair drive prompt, try again, but keep receiving the same message, only after attaching the USB to the SmartBench. 

https://yetitool.atlassian.net/browse/CI-3

LettyYeti, 2020-04-08:
@AzYeti @Neex101 I expect this to be solved by the changes to USB reliability, but obvs needs confirmation.

Neex101, 2020-04-09:
Cheers will test... @AzYeti you got any repeatable fail procedures from before which I can test?

Neex101, 2020-04-14:
@AzYeti you still seeing issues since latest master?

AzYeti, 2020-04-14:
Haven’t tested enough yet. Need to get a customer onto master to test but their closed

 

Thanks,

Aaron

AzYeti, 2020-07-17:
Yes,  I had a customer with this issue yesterday

 

@AzYeti <https://github.com/AzYeti>  any updates on this? Is it still a problem?

closed 2020-07-31 by LettyYeti