YetiTool/easycut-smartbench #223: tap on the ./ line in usbstick load file crash everything. need reboot

sergevi, 2019-12-02:
I often tap a little too high on the file manager in the USB file chooser and choose the "./" line.
I immediately get the yetitool spashscreen and impossible to come back.
Need to stop and restart all.

Neex101, 2020-04-01:
@sergevi thanks for logging this, we're scheduling a fix for our next milestone release.
Annoyingly this is a funciton which is quite deep in the kivy framework, but we'll have a go!

sergevi, 2020-04-01:
you welcome. It doesn't prevent to work at all  :) I pay attention now and all is good.. happy you will debug it :)

Neex101, 2020-04-01:
We'll get there!

Neex101, 2020-04-02:
@Lettie, from memory, I think switching to `iconView` as the filechooser template might actually get rid of this problem - weird, I know, but lets try

Neex101, 2020-04-12:
@AzYeti pls confirm this is now fixed

Neex101, 2020-04-14:
Thanks @sergevi this is now fixed and ready for next release (due next week :-)

PR #367 (2020-04-14): Filechooser improvements: enabled folders, filetype info and dodged crash ./ line
- Enabled local & USB folder visibility: Closes #366 
- Removed ./ line in usbstick load file dialogue: Closes #223 
- Indicated filetypes shown: Closes #214
closed 2020-04-14 by Neex101