YetiTool/easycut-smartbench #409: Analyse GCode to catch high feedrates during GCode check

Neex101, 2020-04-30:


LettyYeti, 2020-05-04:
need a warning in the gcode check

LettyYeti, 2020-07-30:
@Neex101  what counts as "high" ?

Neex101, 2020-07-30:
Hey! Pls check in with Az on this.
Also, this should be retitled to include high spindle RPMs.

-- 
S i S
*Ed Sells* • *Technical Director*
T • +44 1275 217060  M • +44 7833 478878
E • ed.sells@yetitool.com  W • www.yetitool.com
A • Unit 1 Westfield Court | Barns Ground | Clevedon | Bristol | BS21 6FQ |
UK
[image: Follow us on Facebook.] <https://www.facebook.com/yetitool/> [image:
Follow me us Twitter!] <https://twitter.com/yetitool> [image: Follow us on
LinkedIn!]  <https://www.linkedin.com/company/yeti-tool-ltd/>[image: See
our Yeti TV channel on YouTube.]
<https://www.youtube.com/channel/UC9dWz-Uk1K3ewhxLDG7yjtQ>
The e-mail and any files transmitted with it are confidential and may be
privileged, and are intended solely for the use of the individual or entity
to whom they are addressed. Any unauthorised direct or indirect
dissemination, distribution or copying of this message and any attachments
is strictly prohibited.

LettyYeti, 2020-07-30:
@AzYeti help plz

AzYeti, 2020-07-30:
Hi?

LettyYeti, 2020-07-31:
migrated to trello

closed 2020-07-31 by LettyYeti