YetiTool/easycut-smartbench #426: Handle non-conforming files

Neex101, 2020-07-16:
Currently a dodgy GCode file can crash EC.
Example, try swapping a full-stop for a comma in any file.
"The same happens when you choose the wrong post-processor in your CAM-Tool" (- user)

We need to bomb to a "problem with your file" screen, not the blue Yeti logo.

LettyYeti, 2020-07-17:
Where does it crash? Is this if it is/isn't checked during a cut?

Neex101, 2020-07-17:
Don't know, just try a file with a comma in a value

LettyYeti, 2020-07-24:
@Neex101 I've tried this and am just getting standard Gcode errors up, easycut isn't falling over.

Do you have any examples files?

Neex101, 2020-07-24:
I will track down the source

-- 
S i S
*Ed Sells* • *Technical Director*
T • +44 1275 217060  M • +44 7833 478878
E • ed.sells@yetitool.com  W • www.yetitool.com
A • Unit 1 Westfield Court | Barns Ground | Clevedon | Bristol | BS21 6FQ |
UK
[image: Follow us on Facebook.] <https://www.facebook.com/yetitool/> [image:
Follow me us Twitter!] <https://twitter.com/yetitool> [image: Follow us on
LinkedIn!]  <https://www.linkedin.com/company/yeti-tool-ltd/>[image: See
our Yeti TV channel on YouTube.]
<https://www.youtube.com/channel/UC9dWz-Uk1K3ewhxLDG7yjtQ>
The e-mail and any files transmitted with it are confidential and may be
privileged, and are intended solely for the use of the individual or entity
to whom they are addressed. Any unauthorised direct or indirect
dissemination, distribution or copying of this message and any attachments
is strictly prohibited.

closed 2020-07-29 by LettyYeti